marionette-daq/marionette-hardware #7: J7 should be a no place component

jalcok1, 2015-06-04:
If a user adds a shorting jumper across J7, 3.3V is shorted to ground and marionette reboots.  I believe the original intent was not to place this but we did not document it anywhere.  

If a user wants to provide a different Vref for the DAC they need to replace R18 with a different value, right now it is a 0 ohm resistor.

andrewgreenberg, 2015-06-04:
Hi!

Yes, that's probably a good idea. R18 and J7 should never be used at the
same time - you should desolder R18 and then add J7 to go to your own
custom vref. Hardly anyone will ever do this, so we don't need to
populate J7.

> If a user wants to provide a different Vref for the DAC they need to
> replace R18 with a different value, right now it is a 0 ohm resistor.

Again, they'll actually have to remove it.

Andrew

-- 
---------------------------------------------------------------------------
       Andrew Greenberg - andrew1602@gmail.com - 503.708.7711
---------------------------------------------------------------------------

andrewgreenberg, 2016-06-28:
Now is a no place component.

closed 2016-06-28 by andrewgreenberg