marionette-daq/marionette-hardware #13: Resetting the external DAQ

kwilsonpdx, 2015-09-08:
### Found this note from the V2 Capstone guys. Do we want to implement a hardware reset?

* Maybe gate a GPIO with Board RESET_ and Supply...i dunno. More discussion needed?

*"If the dac124 chip has been configured and outputs have been written it will not reset if the marionette board is reset via software or the reset button. As it is powered by V CC , the registers on the dac124 will continue to output until it is reset or the board is removed from its power source. A possible fix to this would be to use a GPIO pin as the 3.3V power source so that when the board is reset and all pins are set to input floating the dac chip powers down."*

### Datasheet:
https://github.com/marionette-daq/marionette-hardware/blob/master/Datasheets/dac124s085.pdf

andrewgreenberg, 2015-09-08:
Oh hey, this is important. A user would expect a reset to reset EVERYTHING, most importantly any outputs. So let's do this. Except instead of powering the chip off the GPIO, just have it control a P-CH MOSFET to the analog supply.

andrewgreenberg, 2016-06-28:
Oh. Shit. Yeah, we forgot this on v2.1. WHY DOES NO READ THE ISSUES?! I keep telling... uh... telling myself.... uh... yeah.

Scheduling for v2.2... :/

andrewgreenberg, 2018-03-29:
Fixed! NRST is now hooked to V3.3DAC LDO's "EN" input. When NRST is pulled low, then v3.3DAC turns off, which power cycles and thus resets the DAC.

closed 2018-03-29 by andrewgreenberg